zeus074/dramtester #1: .INO file for Arduino IDE

repetto74, 2021-09-11:
Hi,
I am building this very nice project. 
I used to program my Arduino boards with the Arduino IDE but it requires files with .ino extension (while yours is a compiled hex).
Any chance to add also the .ino file to load on the Arduino IDE?
Many thanks.
Rick

zeus074, 2021-09-13:
Hi, yes the .ino file is the source code and is still under development, so I had put the hex and update it from time to time. if you want you can also load the hex with the Arduino compiler, you have to go from the command line to the folder where the software is installed and launch avrdude.
ES:
C: \ Users \ "Your username" \ AppData \ Local \ Arduino15 \ packages \ arduino \ tools \ avrdude \ 6.3.0-arduino17 / bin / avrdude -CC: \ Users \ "your username" \ AppData \ Local \ Arduino15 \ packages \ arduino \ tools \ avrdude \ 6.3.0-arduino17 / etc / avrdude.conf -v -patmega328p -carduino -PCOM5 -b115200 -D -Uflash: w: C: \ Path hex file "win-temp \ arduino_build_678219 / Dram -lcd_v4.2.ino.hex-tester: i

Alternatively you can simply download an ide for avrdude and launch the firmware from there:
https://blog.zakkemble.net/avrdudess-a-gui-for-avrdude/

repetto74, 2021-09-13:
Hi Mirco,

Thank you for your time :-)
I finally tried AVRDUDESS on a Windows machine (I am on MacOSX) and I could
load the hex pretty easily on the Nano :-)
Still waiting for the PCB to build this nice project.
I assume that the DC DC booster converter needs to be set to +12v (with the
variable resistor) as those will be used by the 4116s ?
Best regards,
Riccardo

zeus074, 2021-09-13:
Yes the dc-dc converter must be set to 12v. for dram 4116s i have never tried them but i think it works.
Regards.

Mirco.

closed 2021-09-13 by zeus074

repetto74, 2021-09-30:
Ciao Mirco :-)

Ho capito che sei italiano, hai un canale Youtube.
Ho montato il tuo tester DRAM e funziona benissimo con le 4164 :-).
Riscontro pero' un problema per le 4116. Ho cercato di regolare il booster
a +12v in uscita ma niente quando giro il potenziometro il voltaggio non si
muove sul multimetro e rimane a 3.9v in uscita. In entrata ho 4v precisi il
che sembra bassino mentre sull'Arduino mini misuro 4.6v.
Non capisco dove stia sbagliando nel cercare di regolare il booster.
Possibile che sia fallato? Ne ho altri di ricambio ma forse prima dovrei
controllare altro?

PS : I jumper sono stati posizionati correttamente sulle 4116.

Grazie in anticipo per il tuo eventuale consiglio :-)
Riccardo

zeus074, 2021-09-30:
Ciao Ricardo, il booster sarebbe meglio se regolato prima di metterlo nel circuito così sei sicuro del suo funzionamento, ma non è sicuramente questo il problema. La 12V è usata solo nell selettore - pin del 4116, quindi da escludere un abbassamentp da assorbimento anomalo.
Controlla che il booster accetti una tensione di 4-5v in ingresso, alcuni funzionano solo con tensioni più elevate.
Controlla anche i pin di ingresso che magari non fosse stato collegato al contrario se non ci sono serigrafie ad indicare in - out.
Se tutto questo è corretto allora non funziona il dc-dc.
il dc-dc che ho usato io è questo:
https://www.amazon.it/dp/B07MY3NZ18/ref=cm_sw_em_r_mt_dp_EECSXGNEZ534HXSNFZWJ?_encoding=UTF8&psc=1

repetto74, 2021-10-01:
Ciao Mirco,

Ho preso questo booster per 4 lire su Aliexpress pensando fosse una sola
:-). Ma vedi a non sapere bene come usarlo si pensa male.
Devo dare parecchi giri al trimmer per iniziare a vedere un cambiamento dei
valori delle tensioni. Dopo 10 giri mi fermavo pensando che non funzionasse
invece bisogna dare almeno 20 giri e più per regolarlo :-). Adesso funziona
con +12v precisi in uscita sul multimetro :-).
Ottimi progetto mi mancava questo tester RAM quando aggiusto le PCB dei C64
e macchine Spectrum :-)
Grazie a te!
Riccardo