zeus074/dramtester #5: Roughly -4V instead of -5V

seanch80, 2022-05-01:
Hello.  Taking the 5V (more realistically 4.7V if powered via USB) from the Arduino +5V pin and routing it through the diode only gives about ~4V into the 7660S, giving a resulting negative voltage of about -4V. According to the datasheet of a randomly chosen 4116 DRAM chip, the min/max acceptable values for the -5V line is -4.5V to -5.5V.

How does that work? Are typical 4116 DRAM chips not so picky when it comes to the -5V line?

Also, I wonder why the 1N4148 diode was chosen, and not a schottky diode or other with less voltage drop?

zeus074, 2022-05-02:
Yes, you are right, the voltage could be a bit at the limit. I have seen from the tests that it works ,but to improve it without modifying the scheme, just replace the 1n4148 diode with a Schottky one such as BAT43 or similar. 
It is also possible to remove the diode and put a jumper, this was left over from the old version where I supplied the 12V externally.

seanch80, 2022-05-02:
Thanks for the explanation. I think I will use a jumper instead of the diode then.

closed 2022-05-02 by seanch80

seanch80, 2022-05-08:
[image]

..with the diode replaced by a jumper. 😄

seanch80, 2022-05-08:
[DRAMTesterBase_01.zip](https://github.com/zeus074/dramtester/files/8646154/DRAMTesterBase_01.zip)

STL file of 3D printed base.

[image]

zeus074, 2022-05-08:
good job!
I post the file on Github with thanks. 
I also added the stl for print the handle to glue all the jumpers and move them more easily between the high and low position.
👍

seanch80, 2022-05-09:
Very nice. Thanks for sharing this project on github. It was fun to build and works great!