zeus074/dramtester #6: QUESTION: Can I use your kit to test these?

iosoft, 2022-08-31:
Can I use your kit to test these?
KM41256AP-12
KM41464AP-10
etc.

[KM41256AP-12.pdf](https://github.com/zeus074/dramtester/files/9463470/KM41256AP-12.pdf)
[KM41464AP-10.pdf](https://github.com/zeus074/dramtester/files/9463471/KM41464AP-10.pdf)

zeus074, 2022-10-15:
hello, I don't know, I've never tried them but I'm afraid that the page/nibble can give problems in reading and writing with the classic 41256 system.

giobbino, 2022-11-26:
> Can I use your kit to test these? KM41256AP-12 KM41464AP-10 etc.

In my case it worked with KM4164B-15. 

I tested WCM41256FX-20R and they passed the test (actually I have some doubt about these RAM ICs, since in the past I marked them as "probably bad" and they passed the test).