zeus074/dramtester #8: No 12V output, only about 5V through the MT3608

feltel, 2022-10-22:
I just built a DRAM tester with your PCB. Thanks for sharing. I came across an issue with the 12V DC-converter. If I connect the tester the proper way through the Arduino I get only about 5V out of the MT3608 converter. So I cannot use it to test 4116 RAM. If I feed the tester via a bench power supply (set to 5V/1A) and connect it to the VIN terminals I get the desired voltage of 12V at the VOUT terminals. I don get it. There is not much between the Arduino power output pin and the VIN terminal. I double and tripple checked the resistor values and everything else. I used an resistor lead instead of the diode at D1 as discussed in another issue here. Am I missing something?

zeus074, 2022-10-22:
Hi, the 5V is taken from Arduino pin 27 and powers the 7660 display and the step-up as well as the ram.
The diode was giving problems to the 7660 because it could lower the voltage too much.
You can try to check the input voltages at the step-up or put a temporary wire up to pin 27 of the Arduino.
Alternatively, try to use the diode to power only the step-up, to eliminate any return disturbances.
Otherwise I think the problem may be Arduino or the step-up.

feltel, 2022-10-24:
Dear Mirco,
I replaced the wire at D1 with an diode as per the BOM. The voltage at the 12V rail dropped as expected even more. Its now at around 3,9V. The arduino outputs about 5V at Pin 17. As next test I soldered an patch wire from the Arduino socket, Pin 17 to the VIN+ MT3608 at the underside of the board. No change in voltage. I moved the patch wire to the top side of the MT3608 (just to rule out a non optimal connection), but no change at all. I switched the arduino to another one, but also no change. I have flashed two arduinos, one from AliExpress (due to USB-C connector) and one from an "trusted" local source. But both arduinos show the same behaviour. I'm out of ideas and cannot explain, why I get the expected voltage if I use an bench power supply connected to VIN+, but not with the arduino.

giobbino, 2022-11-26:
feltel, I my case it works fine. I set the DC-DC converter at 12v before to solder it on the board, and now it's supplying 12v when powered by Arduino.

I should add that I'm not using an Arduino Nano; instead, I'm using a drop-in replacement, the Mini-EVB LGT8F328P-LQFP32 a 32bit board that runs at 32MHz... but it shouldn't make any difference about that.

I bought the DC-DC converter on Aliexpress (not the most reliable source, indeed) but it works fine. Are you sure you haven't some issue with your DC-DC converter?

feltel, 2022-12-02:
I'm not sure regarding the converters. But why does it work when "jump starting" directly from the converter. I have some converters from another source on back order and will give it a try asap.