zeus074/dramtester #10: RAM chips that didn't work in the real life are detected as OK

giobbino, 2022-11-25:
I can't say if it's an issue or not...  I'm just wondering if there isn't a way to add some more exhaustive tests.

I have few ICs that I marked as possibly bad, since I tested in my computers (C=64, etc.) and they didn't work. All of them are part of a bigger lot; most of them worked fine, some not. I would exclude a compatibility issue for that reason.

Most of the bad ones were correctly detected by the RAM tester, but few of them (3x 4164 and 2x 41256) passed the test.

Actually I don't know if they're ok or the RAM tester makes just a weak test that passes inconclusive results. 

Any idea, any way to add a "stress test" to the ICs?

Cheers,
Giovi

zeus074, 2022-11-26:
Yes, unfortunately the test is not 100% reliable also because the cell is tested only once per test and the timing is not precise. I might could try to see if 2-3 readings of the same cell give different results. Unfortunately I don't have a faulty memory bank on the computer but it passes the test successfully. when I check the voltages on the divider I try to add a second reading

giobbino, 2022-11-26:
Mirco, I can make all the tests for you, if you want... I have many faulty chips here LOL
ciao
Giovi