zeus074/dramtester #9: Jumpers have no effect (doesn't change between 4116 and 4165/256)

giobbino, 2022-11-25:
This is quite a trivial issue, but since it took a little to understand what was wrong, I just want to leave a clue (also because some note inside the code are in italian... perfect for me, but not so good for somebody LOL).

Issue: no effect moving the three jumpers on High or Low. 

Looking at the source code, in the function.ino -> selchip() there's a static declaration about the three values: off, 4116 and 4164 from the A6 pin: below 10 is OFF, between 10 and 200 is 4164, otherwise is 4116.

In my configuration, the setting for 4164 returns about 720-730 and for 4116 is above 1800.

So if you experienced the same problem, you should open the source code, find the selchip() function  and activate the line:

   Serial.println(MREAD);

Then compile, upload and run the code on your RAM tester, and open the serial monitor with Ctrl+Shift+M ( ** CHECK THE SPEED IS 9600!)

Check the three jumpers setup: no jumper, 4164 and 4116.

Then come back in the selchip() function and fix the lines with the correct values:

if (MREAD<10) {//4164     <<< put a value a little above the "no jumper" config. if it is zero, this line is already ok
      return 0;
} else if (MREAD<200) {//4164-256-32   <<put a value a little above the "2164 jumper" config. in my case was ~730 so I put 800

This should fix the issue.

------------------------
I have to say I used a "nano-clone", drop-in replacement board, the Mini-EVB LGT8F328P-LQFP32 @ 32MHz. This board should be an almost perfect clone of an Arduino Nano, but I can't 100% guarantee about that.

Cheers,
Giovi

zeus074, 2022-11-26:
this may depend on the supply voltage and resistance values, I calibrated them according to the settings I had. to avoid problems, the intervention range can be increased. check with mine and in case I increase the value.
I got these values: (when I write the code)
0=key on 4164 510=empty key 1024=key on 4116

1800 seems like a very high value to me.

giobbino, 2022-11-26:
I think you mean 0=empty and 510=key on 4164 (your comments in the code seems to be different from the source code)

I got: 0=empty; 720~730=4164; 1815~1817=4116. 

I could depend by the different board (Mini-EVB LGT8F328P-LQFP32) I used, but I'm not really sure about it, since I used that board in many projects (it was 1/2 the price of an Arduino Nano clone so I bought some of them) and I never experienced any issue. However, no problem at all, just a matter of setting the value in the source code.

To make the things simpler, I slightly modify your code:

I added two lines at the top of the main sketch...
#define RAM_NDEF  10    // <   10 = empty
#define RAM_4164  1000  // < 1000 = 4164    

...and I modify the selchip() function accordingly 
    if (MREAD<RAM_NDEF) {  //no jumpers
      return 0;
    } else if (MREAD<RAM_4164) { //jumpers on 4164
      return 1;
    } else {  //jumpers on 4116
      return 2;

giobbino, 2022-12-02:
Thank you. Soon or later i will build another with an Arduino Nano, just to see if the problem is the Arduino Nano vs clone or the resistors. However, generally speaking, I see no needs to change the hardware when you can solve changing the software LOL...
With the new values it's working, and it's enough for me.

I also slightly modify you software, adding a FAST / SLOW mode on all modes. The SLOW mode makes more test per bit, writing 0 and 1 alternately. However I didn't see any difference on the RAM chips that were supposed to be bad , so I don't know if there's any good doing that.

Next step, is to test those "bad" (bad?) RAM ICs again in a C=64 and see if they are really bad or not.

giobbino, 2022-12-03:
I've built a second RAM tester, this time using a chinese Arduino Nano. All the others components remain the same, from the same seller and lot.

The values changed dramatically: now I have a value around 185 for the 64/256 setting, and around 473 for the 4116 setup.

Weird, isn't it?